MoffKalast/vizanti #9: ROS 2 support?

swarajpeppermint, 2023-05-31:
Does this support ROS 2?

MoffKalast, 2023-05-31:
Right now it's Noetic only (possibly Melodic if compiled with python3, though untested) @b1n-ch1kn is taking a crack at a ROS 2 port: https://github.com/b1n-ch1kn/vizanti though it's not completely working yet.

MoffKalast, 2023-06-23:
Ok so it might be useful to continue the discussion here, from https://github.com/MoffKalast/vizanti/pull/15, since the build is partially working now, on branch ros2.

The list of things tested and fixed, and those left either untested or still requiring work:
- [x] Global Settings
- [x] Grid
- [x] TF
- [x] Robot Model
- [ ] Dynamic Reconfigure (needs partial rework, disabled)
- [ ] Bag Recorder (won't report correct recording status at times, disabled)
- [x] Node Manager (a bit iffy in terms of finding executables and killing nodes)
- [x] Add new visualizer/widget (though seems to have higher latency in finding new and discarding old topics)

---

- [x] Teleop Joystick (tested with turtlesim)
- [ ] 2D Pose Estimate
- [ ] 2D Nav Goal
- [ ] Waypoint Mission
- [ ] Area Mission
- [x] Button (tested Empty with vrx ball launcher)

---

- [x] Map (tested with dummy_robot_bringup)
- [x] Satellite Tiles (tested with vrx)
- [ ] Battery
- [ ] Compressed Image
- [ ] Marker Array
- [ ] Path
- [ ] Range
- [x] LaserScan (tested with dummy_robot_bringup)
- [ ] PoseWithCovarianceStamped
- [ ] Pose Array
- [ ] Temperature

@swarajpeppermint if you've got a robot that has any of the unchecked topics above to visualize it would be a great help to try and confirm what works and what doesn't. I'm still struggling to find good ROS 2 demos in gazebo or rosbags that would test the bulk of it.

MoffKalast, 2023-06-25:
Ok got the nav2 demo working to some extent, the Header seq param no longer exists, so that was throwing errors. That should fix initialpose and simple goals.

https://github.com/MoffKalast/vizanti/assets/9977799/d9cc6fb3-d447-433c-8651-4905c02d6d73

swarajpeppermint, 2023-07-04:
@MoffKalast sure. It looks like an interesting project that we can check/use in our organization internally (and probably even contribute to).
I've shown this issue to another Robotics Engineer in the team. If we decide to explore this, we'll definitely get back to you.

MoffKalast, 2023-08-26:
Alright this seems more or less done now, with #27 merged.

closed 2023-08-26 by MoffKalast

factoidforrest, 2024-09-10:
What are the setup steps to use vizanti with ros 2? I am somewhat new to the ecosystem so I am stumbling on the inconsistencies between versions in the installation guide for vizanti. It seems to depend on catkin, for instance

factoidforrest, 2024-09-10:
Oh, I see. I managed to find the fork:

https://github.com/MoffKalast/vizanti/tree/ros2

My appologies if I missed any mention of that in the main readme. If there aren't any, I might add one if I do any other PRs!

MoffKalast, 2024-09-10:
Yep, you have to look at the ROS2 readme for installation instructions: https://github.com/MoffKalast/vizanti/blob/ros2/ReadMe.md

It's a fairly standard ROS thing to have separate branches with different documentation pertaining to that specific version, but I see it would be wise to add a note about it on the wiki for newcomers. I hadn't really anticipated that this repo would ever be looked at by people completely new to ROS itself :sweat_smile: 

ROS 1 Noetic is still the main branch since it's the most stable, ROS 2 Humble works mostly reliably with rosbridge as the backend but it is slower, that's why the RWS option is there which is very fast but still has a handful of bugs that need to be looked at. AS for ROS 2 Jazzy, the last I checked (a few months ago) the package builds fine but neither of the backends worked yet, so I'm not sure how well it runs yet.

factoidforrest, 2024-09-17:
Thank you very much for the information! That is certainly on me for trying to learn ROS and Vizanti simultaneously, my appologies. I didn't realize it was a normal convention. 

Thank you for the compatibility hints as well. My platform is a pi 5, and I believe that forces me onto Iron at the minimum. I may just have to containerize humble, always an option.

MoffKalast, 2024-09-17:
@factoidforrest Ah yeah that is a problem on the Pi 5, they never really released anything older than 23.10 for it, aside from some Armbian builds I think, not sure if they work and they wouldn't be tier 1 supported by ROS anyway.

Anyhow I've got a similar setup planned for an AGV I'm refitting, Pi 5 + 24.04 with Jazzy so I should be able to test out any bugs you find.

I've re-tested the state today and while rosbridge from apt doesn't work yet, if you `git clone https://github.com/RobotWebTools/rosbridge_suite` into `colcon_ws/src` and build it along with the rest of the packages it does seem to work at first glance. I've tested what I've got set up as tests in the demos package:

[image]

It's really laggy though, and rosbridge pins one core to 100%. I'll see if I can help RWS get up to speed sometime soon as well.

MoffKalast, 2024-09-19:
RWS is now [also patched for Jazzy](https://github.com/v-kiniv/rws/pull/32), I'll update the wiki.

PR #31 (2023-08-26): Param reconfigure fixes, nav2 demo
Generally closes #9 but needs more testing.