MoffKalast/vizanti #28: Safari widget closing not working

mspringer1, 2023-08-13:
When adding a new widget or configuring an existing widget in Safari on iOS, tapping on the background to close the setup modals does not work. The only way to get back to the default view is to reload the page, restart the web app or to remove the widget.

Tested under iOS 16.5.1 on an iPhone and iPad.

MoffKalast, 2023-08-13:
Hmm, I think I've heard of this issue before but I don't currently have any devices capable of running Safari on hand to reproduce it. Would you be up for trying a few things to try and pinpoint what's causing it?

Looking at the related css classes, the most likely culprit is the `pointer-events: auto;` line in  [modal_outer](https://github.com/MoffKalast/vizanti/blob/7fb84d9a126250abff22c99cda19c85a1f06c9d1/public/css/style.css#L173), which if removed or set to `none` or `inherit` causes similar behaviour in chrome and firefox (clicking does nothing, the main view gets events instead). It should be supported on webkit as well, but maybe auto doesn't quite work there. Most of the [other options listed here](https://developer.mozilla.org/en-US/docs/Web/CSS/pointer-events) seem to get good results on chrome, so `all`, `fill`, `stroke ` and probably some others would likely be working alternatives if they work on safari. 

If that doesn't do anything, another option might be to try commenting out the lines of [the noselect class ](https://github.com/MoffKalast/vizanti/blob/7fb84d9a126250abff22c99cda19c85a1f06c9d1/public/css/style.css#L464) which prevents selecting text when dragging, the webkit lines might be interfering with touch events.

If neither of these works then we'll have to get more creative I guess 😄

mspringer1, 2023-08-14:
I tried to test the changes but wasn't really successful

Trying pointer-events:

- `pointer-events: all;`: does not make a difference

- `pointer-events: fill;`: does not make a difference

- `pointer-events: stroke;`: does not make a difference

Trying noselect class:

- commenting out the webkit classes does not make a difference but just allows to select the text in the webpage
- commenting out the all classes has the same effect

is there anything else i could test?

MoffKalast, 2023-08-14:
Hmm well I've set up a branch `safari-modal-fix` that removes pointer-events entirely and instead has the modal container element expand on demand and always capture clicks. That ought to work I think, can you give it a test?

mspringer1, 2023-08-14:
I just tested the `safar-modal-fix` branch and unfortunately it is still the same behavior when closing a widget.

I also noticed a transparent circle icon that sometimes appears at the tapped area in safari. Maybe a magnifying glass function.
[image]

MoffKalast, 2023-08-14:
Ok if it really is magnifying glass related, then it may require some kind of preventDefault in the event chain that's calling it by default. I've pushed another change that adds the closing listeners directly onto the elements, and stops any propagation.

MoffKalast, 2023-08-15:
@mspringer1  I've mentioned this issue to a friend of mine who has a mac, and it may turn out to be as simple as changing `window `to `document `in the listener part. If #29 if fixes it we can just go with that instead.

commit d5f6f0ae (2023-08-16): Merge pull request #29 from emir7/fix/modal-click-listener

Closes #28 for noetic
closed 2023-08-16 by MoffKalast