MoffKalast/vizanti #35: Add base URL option

tony2guo, 2023-09-01:
Hi, thanks for the great work!
I would like to use Vizanti under some subfolder like this: http://<host_ip>/vizanti
Possible adding a base URL option or use relative URLs?

MoffKalast, 2023-09-01:
Hmm well if by host_ip you mean without the :port part that's a bit of a problem. ROS really doesn't want anything to do with root and linux won't let anything but root host on port 80, so it's a bit of a standoff. One option that I've used in the past is to set up a root hosted redirect server like nginx on port 80 separately, that then routes to 5000. Or to run `server.py `with `sudo python3`.

The server expects [some specific paths](https://github.com/MoffKalast/vizanti/blob/noetic-devel/src/server.py) which would need to be adjusted on both ends to support the extra /vizanti, but in theory it would be doable.

tony2guo, 2023-09-04:
I am using Traefik as a reverse proxy to route subfolder /vizanti to port 5000 but found some errors due to some URLs is not relative URLs. I'll try to fix this and create a PR.

tony2guo, 2023-09-04:
To add a base URL, I've tried [blueprints](https://flask.palletsprojects.com/en/2.3.x/blueprints/) but can not get it to work.
I've also tried to hardcode the base URL: https://github.com/MoffKalast/vizanti/compare/noetic-devel...tony2guo:vizanti:base-url
It works, but there should be some better ways. I'm not familiar with Flask, so I'm not sure how this should be done.

MoffKalast, 2023-09-05:
Hmm well the current setup uses jinja templating, which seems to have some kind of [url_for function](https://stackoverflow.com/questions/45207969/how-can-i-get-the-base-url-with-flask-jinja2) that would be added to returned templates. Not sure if it's the correct URL though.

tony2guo, 2023-09-07:
Thanks for the info. I'll try url_for().

closed 2023-10-10 by tony2guo