MoffKalast/vizanti #38: Possible bug in get_public_dir

tony2guo, 2023-09-04:
https://github.com/MoffKalast/vizanti/blob/9c5b1b95bf4815c3c5d76d4c2fd4f4c8903568e7/vizanti_server/vizanti_server/server.py#L14
This would only work if server.py is run from vizanti/vizanti_server/vizanti_server:

`~/ros2_ws/src/vizanti/vizanti_server/vizanti_server# ros2 launch vizanti_server vizanti_server.launch.py`
[vizanti_flask_node]: Public directory set as ../public

`~# ros2 launch vizanti_server vizanti_server.launch.py`
[vizanti_flask_node]: Public directory set as /root/ros2_ws/install/vizanti_server/share/vizanti_server/public/

A possible solution:
```python
from pathlib import Path
def get_public_dir():
    p = Path(__file__).resolve()
    path = p.parents[1] / 'public'
    if path.exists():
        return path #for --symlink-install
    return get_package_share_directory('vizanti_server')+ '/public/'
```

MoffKalast, 2023-09-04:
Oh there definitely is a bug there. The general idea was that when someone runs `--symlink-install` the server script would run in its source package dir and find the relative public folder there, but it never quite worked except for the first time I tried it when I was probably cd'ed to the package dir. Interesting that the actual shell command path matters, I wouldn't ever expect that to be actually used. Iirc in ROS 1 it always took the script directory as the execution dir even when using symlinks. 

Anyhow, well spotted, I'll see if I can test it out later today :+1:

MoffKalast, 2023-09-07:
Seems like it works, merged. :+1:

PR #39 (2023-09-07): Public dir lookup fix
Fixes #38
closed 2023-09-07 by MoffKalast