MoffKalast/vizanti #50: Working behind proxy

giovanidiniz, 2023-10-22:
Hi, guys, 

I've been exploring vizanti as a consolidated web interface for my aerial and ground platforms. 

I've been having trouble running SITL instances behind a proxy on my cloud server using the ros2-feature branch. 
The best I could get so far is a screen with background colors, but nothing else. It looks like I can reach the base html but none of the js stuff is reachable, maybe?  

So far, I've tried to use all sorts of values for `base_url` to try and fix this with no success. I'm wondering if I there might a need to set the js files paths (although I thought that's what `url_for` did).

Any ideas would be super welcome, I'm at a loss now

Thanks!

MoffKalast, 2023-10-22:
Hmm well that's peculiar for sure, if you're not seeing any icons in the top bar after loading then either JS is either not being served or is being blocked by the browser. What kind of console/network errors do you get client side, if any?

If your proxy is only set up for specific ports, then you might wanna check if 5000 (static serve) and 5001 (rosbridge) are being forwarded. I suppose you could try a Wireshark capture when accessing it from the same machine and then again through the proxy to see what's not getting through.

giovanidiniz, 2023-10-22:
Ah, I'm getting a bunch of these:
```
my.domain.com/:1 Refused to execute script from 'https://my.domain.com/js/setup.js' because its MIME type ('text/html') is not executable, and strict MIME type checking is enabled.
```
I think you're right, I'm missing the actual proxied route between my.domain.com and the /js stuff.

MoffKalast, 2023-10-23:
That does actually ring a bell, the mime type [is being set explicitly for most files](https://github.com/MoffKalast/vizanti/blob/749d54e55a05c7cc853476813ec17b6986ae6d32/vizanti_server/vizanti_server/server.py#L42), otherwise the scripts can't be imported. It's interesting that it's failing on the statically declared ones. Maybe your proxy is somehow changing mime types in transit? The [declaration of those is here](https://github.com/MoffKalast/vizanti/blob/749d54e55a05c7cc853476813ec17b6986ae6d32/vizanti_server/public/index.html#L26) btw, if you want to try changing it and seeing if anything helps.