MoffKalast/vizanti #70: Map not rendered

kosmonauta144, 2024-04-04:
Hi,
we use Vizanti alongside RTABMAP. The map is published on topic `/rtabmap/map`. The map is not published regularly, but can be triggered with the service. In rviz the map is shown, but it is not in vizanti. Costmaps work perfectly normal. It's strange, because the issue occured about 2-3 weeks ago, when from day to day it stopped working. TFs and other topics are correct, but map is not seen. Do you have any idea what we should change to diagnose what's wrong and how to repair it?

[image]

MoffKalast, 2024-04-05:
Well, two potential problems come to mind. There is an issue I've sometimes seen with rosbridge, where it might process latched topics immediately (when there's no web clients connected yet) and those messages won't get sent to every new client since it only counts when rosbridge subscribes, not the websocket. Then again if the map is published while the web part is connected it should definitely receive it.

There is a rendering bug in Firefox that this sounds familiar to, as maps are rendered with a worker in an offscreen canvas to reduce stutter, but oftentimes nothing ends up rendering at all. It's shouldn't be an issue in Chrome, and I have added a workaround for it a while back, though I'm not entirely certain it's a complete fix.

There are a few things to try:
- check the web console (F12) for any errors
- compare Chrome and Firefox
- if on ROS 2, is there any difference between rosbridge and [rws](https://github.com/MoffKalast/vizanti/tree/ros2?tab=readme-ov-file#optional---experimental-rws-backend)
- there are [two WIP PRs for noetic and humble](https://github.com/MoffKalast/vizanti/pulls) that include some changes to map rendering and caching, you could test those out if the behaviour is any better
- if you can reproduce this in a short rosbag that I could test with locally, that would be most practical

kosmonauta144, 2024-04-08:
We checked the web console, but no errors have occur.
We checked Chrome and Firefox and there is no difference between them. On both the map is not displayed.
I will check rws today, because I have not checked it yet.

Here is the bag with publishing map on the topic and tf's. Vizanti is started after booting the system with crontab, but I don't know if it has an influence.
[vizanti_diagnostic.zip](https://github.com/MoffKalast/vizanti/files/14901726/vizanti_diagnostic.zip)