MoffKalast/vizanti #72: Parameter loading issue

kosmonauta144, 2024-04-16:
Hi,

last time I encontered an issue. I launched some nodes right after booting the system, and they are perfectly fine to set parameters

[image]

imu presented above is only one if few examples. 

But there are some nodes which are launched after startup by user and they are not able to load parameters 
[image]

I don't know if I am doing something wrong, but some parameters are not loading and they are important for me. I can changed them with classic `ros2 param set` but I would prefer to do it through gui. I will be grateful for every hint for fixing this.

MoffKalast, 2024-04-16:
Yeah I'm afraid that's somewhat of a known issue right now. Param reconfigure has been spotty on the ROS 2 branch as a whole.

Currently the node params and their current values are obtained by [running ros2 param dump via subprocess](https://github.com/MoffKalast/vizanti/blob/cc6d6ab09ad24dde7d712f3439f66d66dbe4076f/vizanti_server/scripts/service_handler.py#L241C38-L241C53) with a timeout of 5 seconds. I haven't found this to be reliable even running it manually in the terminal, it's often extremely slow and sometimes it just hangs forever, which may be what you're seeing on the front end side. Or it may be taking 6 seconds and it's always just slightly cut off. You could try increasing the timeout a bit to see if it gets you any closer, but in my tests at the time it wasn't much of an improvement.

It is interesting that rqt_reconfigure manages to very quickly obtain the params and their values, so I presume it's more of a fault in rclpy (since that's what roscli uses now). As such the eventual goal is to[ move the reconfigure part to a cpp node](https://github.com/MoffKalast/vizanti/blob/ros2/vizanti_cpp/src/param_manager.cpp) and use the internal rclcpp SyncParametersClient instead, but that's still on the todo list.

There's also a secondary problem that humble and rolling have some api changes around that internal api, so implementing this now would force another branch split to avoid breaking compatibility, which isn't ideal.

kosmonauta144, 2024-04-17:
thank you very much for your reply. I will try to look into it then

MoffKalast, 2024-10-11:
Addressed by #104 for now.

closed 2024-10-11 by MoffKalast