MoffKalast/vizanti #73: vizanti over virtual network

sualeh-rgt, 2024-04-19:
This is an amazing work, bundle of thanks to the contributers.
Recently I faced an issue and I want to bring it to your attention.

Remote device is running a ros1 system in a virtual network and the intention is to monitor it remotely.

vizanti fails to get the tf data from the remote device and resultantly all tf related topics fail to get subscribed.

BUT

If the vizanti server is launched via ssh then all works fine.

sualeh-rgt, 2024-04-22:
Thankyou for your response!
we tested again on a similar setup, and it worked without any issues.
Since the issue was related to tf only and not the connection itself, I suppose the function "clear_old_tfs" caused the problem.
I will further investigate and let you know.

MoffKalast, 2024-04-22:
Hmm that could be possible, but what clear_old_tfs does is it removes any non-static tf frames that haven't been published for 15 seconds. If you have such frames, it would be best to move them over to /tf_static, otherwise rviz tends to treat them as missing too.

MoffKalast, 2024-05-08:
@sualeh-rgt did you manage to find anything out?

sualeh-rgt, 2024-05-09:
> @sualeh-rgt did you manage to find anything out?

Thankyou for asking! Did not happen again, and I did not explore it further.

MoffKalast, 2024-05-09:
Ah alright, well let me know if you ever run into anything else and I'll close this one for now 😄

closed 2024-05-09 by MoffKalast