MoffKalast/vizanti #78: Does Vizanti have a mobile APK?

EricDaJiang, 2024-05-03:
Does Vizanti have a mobile APK? I saw "The application attempts to replicate RViz's orthographic 2D view as closely as possible with a smartphone friendly interface." mentioned in the text. If so, how do I download, install, and connect?

MoffKalast, 2024-05-03:
[Here's an old video](https://www.youtube.com/watch?v=uXDBWc23ZdQ) of how setup looks on mobile. It doesn't have an APK, but there is a PWA you can make from the mobile browser so it starts as an app in full screen mode. If you follow the readme install setup, it's exactly the same for all client platforms.

It would be great if it were possible to get all the functionality we have from just a frontend. As it is though, it's a two part system and requires a few nodes to run on the ROS side, so the client is also hosted from one of those as a web app. Theoretically one could use Ionic or something to wrap the web part into a PWA deployed as an APK, but it would still need the backend to work and a way to select the correct IP to connect to, which is currently automatic.

EricDaJiang, 2024-05-03:
That is to say, as long as my phone is connected to the same WiFi as the USV, I can open it on my phone http://localhost:5000 , and then set a path point for the USV? Thank you for sharing such an interesting open-source project. I am currently purchasing relevant equipment based on the circuit diagram you provided, and hope to achieve my first USV soon.
Thank you!

MoffKalast, 2024-05-03:
Not exactly, it has to be the USV's IP in that case (localhost works if you're running the ROS part on the same PC as the web client). It really depends on the wifi network. If the USV is hosting a hotspot then something like http://10.24.0.1:5000 or http://192.168.1.1:5000 would be typical, if they're both connected to another network, then you'd have to look up the USV's IP in the router's DHCP UI or set a static one. In any case you'll certainly have to have a way to SSH to it, so using the same IP as for that would work fine.

Np, glad to help 😃

EricDaJiang, 2024-05-03:
One more question, is there only one neo m8n and one antenna installed on the usv? I was about to purchase the neo m8n when I discovered that a neo m8n can only connect to one antenna. There is only one neo m8n on the drawing you provided, and it seems that it usually takes two antennas to calculate the heading Angle?

MoffKalast, 2024-05-03:
Yeah the ones I've used so far are all of [this type](https://external-content.duckduckgo.com/iu/?u=https%3A%2F%2Fstore.nerokas.co.ke%2Fimage%2Fcache%2Fcatalog%2FWireless%2520devices%2FM8N%2520GPS_1-1001x1001.jpg&f=1&nofb=1&ipt=b1653fd1595f1bb8c9902dc7a90100f6d569238d86e914d636d54b4ddf33b6fb&ipo=images) which has one u.fl adapter and one antenna. Getting heading from two would be better though, so definitely go for it if you found one that can do that.

You can sort of get the heading if the receiver is moving fast enough and use a gyro for the quick movements (which is my current implementation for navsat_simple anyway), but it's not ultra reliable.

EricDaJiang, 2024-05-03:
Thanks for the info!