MoffKalast/vizanti #85: Custom Buttons

ValkyrX, 2024-06-09:
Just wondered, is there any way that you could add the ability to send commands to an api or even add the ability to make a button be able to send a command to a php website etc ?

My robot has an api that i can pre-set waypoints etc that can be called by simply calling a webpage and it runs the php code on that page , so is there a way to make the button widget be able to do something like that ?

Thanks
K

Cool interface btw :-)

MoffKalast, 2024-06-10:
Thanks! It might be possible, providing CORS doesn't cause any issues if the API is on HTTPS which iirc would be blocked.

There is also the slight problem that one would need to use XHR or fetch requests for the actual calls, and those are blocked on Safari under HTTP, so it wouldn't work at all on iOS and only on Chrome/Firefox in MacOS.

I think the easier way might be to have an extra backend node that handles this part and isn't constrained by browser shenanigans so it can send the requests when it gets some service call.

ValkyrX, 2024-06-10:
Hmm , so i guess running some kind of node on the ros server that will actually do the api calls would be the way to go ? i wonder if there is one that can be used / manipulated ?

MoffKalast, 2024-06-10:
Yeah I think it's the way to go to get the most done for the least hassle :smile: 

Unless you're planning on using cpp, a simple python node [with flask](https://stackoverflow.com/questions/25149493/how-to-call-another-webservice-api-from-flask) should be all you need.

I'm not sure if anything like that exists already, but here's another chatgpt example as a rough starting point:
```python
#!/usr/bin/env python

import rospy
from std_msgs.msg import Empty
import requests
from flask import Flask, jsonify

app = Flask(__name__)

# URL of the REST API endpoint
API_URL = "https://api.example.com/data"

data_cache = None

def fetch_data():
    global data_cache
    try:
        response = requests.get(API_URL)
        response.raise_for_status()
        data_cache = response.json()
        rospy.loginfo("Data fetched successfully: %s", data_cache)
    except requests.exceptions.RequestException as e:
        rospy.logerr("Failed to fetch data: %s", e)
        data_cache = None

def callback(msg):
    rospy.loginfo("Received a message on the Empty topic, fetching data...")
    fetch_data()

@app.route('/data', methods=['GET'])
def get_data():
    if data_cache is not None:
        return jsonify(data_cache)
    else:
        return jsonify({"error": "No data available"}), 500

if __name__ == '__main__':
    rospy.init_node('flask_fetch_node')

    # ROS subscriber
    rospy.Subscriber('fetch_topic', Empty, callback)

    # Run Flask app in a separate thread
    from threading import Thread
    flask_thread = Thread(target=app.run, kwargs={'host': '0.0.0.0', 'port': 5000})
    flask_thread.start()

    rospy.spin()
```
You'd need to subscribe to [one of these](https://github.com/MoffKalast/vizanti/wiki/Feature-List#mission-planning) and then make the call once you get the message.

ValkyrX, 2025-02-06:
This doesn't work very well for some reason, all I nee the button to do is push api call to a website I can then tell it what to say for example when click button it simply publishes out an api call, each button can have. adiferent function, eg to dock robot I simply need to call 

http://0.0.0.0/api-robo/dock
When my robot gets this it auto docks, does that make sense ?

closed 2025-02-14 by ValkyrX