MoffKalast/vizanti #92: Tilt the map

jalemanb, 2024-07-10:
Is there any plan on adding the tilt capability so one can also change the orientiation of the map, apart from just pan and zoom? Thank in advanced, best regards.

MoffKalast, 2024-07-10:
Ah you mean like yaw rotating the fixed frame around Z? Yeah that's been something I've been meaning to add for a while, but there are a few known blockers and probably a few unknown ones.

The grid widget would need to be fixed so it can render in any frame first (since right now it assumes it can just draw vertical and horizontal lines on the screen), and the area mission widget would likely need some additional handling to account for that rotation. There may be a few other places where zero rotation is assumed for the origin.

But yeah otherwise it would be great to have, multitouch rotation for mobile and something like ctrl+mousedrag or shift+mousedrag for desktop to rotate. Overall it would need a significant amount of testing to make sure it doesn't add any weird rendering bugs since it changes the core foundation of how things are projected.

jalemanb, 2024-07-10:
Yeah this is exactly what I need for an HRI interaction setup, specially since the user is supposed to interact with a touch screen device to visualize the position of the robot and other instances within the map. Thanks a lot for your prompt response, best regards.

MoffKalast, 2025-10-05:
Hey @adamzr2000, there have been some improvements since, mainly #156 that corrected a lot of rendering assumptions so practically all visualizers should handle any set of transforms correctly.

There is still the matter of fixing the grid renderer which renders with the most extravagant set of assumptions for optimization reasons, but after that blocker is fixed it should be possible to add something of the sort by rendering relative to a virtual parent frame to the fixed frame, and then rotate that frame based on user input. 

With that approach, not only yaw but also pitch and roll become possible for an isometric mode, though that currently breaks mouse picking for goal poses and the like (one would need to do raycasting to the fixed frame xy plane instead of using what are essentially just scaled and offset screen coords). At the moment it's possible to preview how that could look like by publishing that virtual frame in ROS instead, e.g. `ros2 run tf2_ros static_transform_publisher 0 0 0 0.7854 0 0.7854 <your fixed frame> view`

So definitely something worth adding, though it's not super high on the list of priorities on my end.