MoffKalast/vizanti #111: how to use "Waypoint Mission"?

dpluy, 2024-10-30:
[Problem description]
I tried this web tool and encountered issues while using Waypoint. After adding multiple points, I don't know how to make the robot start navigating through these points. I tried 'send it', but it didn't work. I haven't seen the relevant operation instructions and am a bit confused. I hope someone can help me.
[image]

**Server Platform(s):**
 - OS: [e.g. Ubuntu 22.04]
 - Architecture: [e.g. x86_64]
 - ROS Version: [e.g. Humble]
 - Branch/Tag: [e.g. ros2]

**Client Platform(s):**
 - OS: [e.g. Android 14]
 - Architecture: [e.g. aarch64]
 - Browser: [e.g. Chrome]

dpluy, 2024-10-30:
Does the image below have a function button to start executing the task
[image]

MoffKalast, 2024-10-30:
Hmm, well the frames seem to be set up correctly, so it should definitely be sending the message when you press "Send it". Do you get anything if you `ros2 topic echo /waypoints`? Do you get any errors if you open the console (F12)?

A few things to note:
- if "Start from closest waypoint" is checked, then the grayed out parts of the path won't get sent, so you should probably disable that if you want the entire path to be sent at all times
- the message sent is a Path or a PoseArray, neither of which are directly used by most nav stacks natively and typically require some kind of manager node that subscribes to that and sends goals one by one to the planner, checking if the previous one has been reached, etc. there's been [some confusion about that](https://github.com/MoffKalast/vizanti/issues/86) in the past anyway

P.S. Your fonts render in a really weird way, what is this running on? The map looks rather odd too.

dpluy, 2024-10-30:
Yes, through topic printing, I received a set of pose_array coordinates, so 'send it' has completed its task.

Based on your second point in response, I may need to do some more work to send this set of coordinates to the planner. Okay, I thought the Nav2 heap could directly receive the "send it" command to complete the waypoints task. I am still a beginner and just started using ros2.

Thank you for your reply.

p.s. 
1. The map in the picture is a map of the real environment created through ros2.
2. I used EDGE browser, which should be similar to Chrome.

MoffKalast, 2024-10-30:
Well for Nav2 there is actually [NavigateThroughPoses](https://docs.nav2.org/behavior_trees/trees/nav_through_poses_recovery.html), and there is a node that packs up a path message for it under vizanti_demos: 
```bash
ros2 run vizanti_demos path_to_nav2poses.py
```

Given that this is a somewhat common pitfall, maybe that node should be set up to launch by default :smiley: 

From what I've tested though, the local planner seems to behave a bit oddly when given this type of path command, as it doesn't really properly reach intermediate goals but just sort of drives through the general area. Probably something that can be configured in Nav2 somewhere, but I haven't really taken much of a deep dive into it either, so I can't yet say for sure.

> The map in the picture is a map of the real environment created through ros2

Well I'm mainly asking since it looks a bit like [a rendering error we had a while back](https://github.com/MoffKalast/vizanti/issues/65), which ought to be fixed now.

>  EDGE browser

Ah interesting, I've never tested it on that but it should be Chromium based yep.

dpluy, 2024-10-31:
I tried this demo
```ros2 run vizanti_demos path_to_nav2poses.py```

It can indeed be achieved. But after completing the complete path once, when I click "send it" again, the robot will not repeat the previous path. There are indeed some strange behaviors, I will continue to try and try to modify them.

Thank you for your work.

p. I tried adding 3 layers and didn't find anything strange, everything looks normal.
[image]

MoffKalast, 2024-10-31:
> It can indeed be achieved. But after completing the complete path once, when I click "send it" again, the robot will not repeat the previous path. There are indeed some strange behaviors, I will continue to try and try to modify them.

That sounds somewhat familiar yes. Well if you find any good fixes let me know and I'll see if we can implement them. 👍

I think given that the waypoint path should generally define the nearly exact way the robot should drive, using navigatethroughposes is probably not the exactly right abstraction. This would essentially need to just be treated as a global plan on top of which the local planner can plot around obstacles, but still sticking to the lines as much as possible. It's something I intend to look into properly once I get around to porting my Noetic USV planner for that, but I haven't gotten around to it yet.

> I tried adding 3 layers and didn't find anything strange, everything looks normal.

Alright, if it looks the same as in rviz then it's probably just the map publisher doing something odd with unexplored pixels.

dpluy, 2025-02-11:
@EricDaJiang Sorry for not responding in a timely manner.

You need to long press the button. If it doesn't work, there may be a bug, right

MoffKalast, 2025-02-11:
Yeah I've explained it to Eric in https://github.com/MoffKalast/vizanti/issues/127 at the time, there may be some potential in adding a double-click-to-open trigger as well in case people get stuck. For now I've just made it a bit clearer in the wiki 😄 

Btw @dpluy did you have any success with making path following more reliable in nav2? I've been mostly busy with a noetic based project for the past months so I haven't really looked into to it myself yet.