MoffKalast/vizanti #122: Visualize the turtlebot4 lite map using namespacing

dkavroulakis, 2025-01-05:
Hello, I would like to visualize the turtlebot4 lite in a web interface and I am using this project: [Vizanti - Web Visualizer & Mission Planner for ROS](https://github.com/MoffKalast/vizanti/tree/ros2) because I can't visualize it using rviz web in ROS2. When I use namespace the map doesn't show up in the vizanti project, but when I remove the namespace the map shows up. 
How could I solve this issue ?
(I have exactly the same issue with y2d2 https://github.com/turtlebot/turtlebot4_simulator/issues/48 and I did everything they mentioned, but the issue still exists.)

**Server Platform(s):**
 - OS: [Ubuntu 22.04]
 - Architecture: [x86_64]
 - ROS Version: [Humble]
 - Branch/Tag: [ros2]

MoffKalast, 2025-01-05:
Hmm I think your issue is that /tf and /tf_static are global topics that can't really be namespaced. Practically all visualizers rely on that and why tf2_tools can't see anything.

I think Clearpath messed up that part, since the way it's supposed to work is to still use the same topics for transforms, but namespace the transforms themselves, e.g. `frame: "base_link"` becomes `frame: "robot1/base_link"`, but still published to /tf. This is necessary if you ever want to transform data from one robot to another with a single tf buffer.

MoffKalast, 2025-01-05:
Yeah seems like that's the thing:

[image]

The good news is that they did do the link namespacing at least:

```bash
transforms:
- header:
    stamp:
      sec: 3
      nanosec: 783000000
    frame_id: /robot1/odom
  child_frame_id: /robot1/base_link
  transform:
    translation:
      x: -2.346575346237305e-12
      y: 2.717228756134409e-25
      z: 0.0
    rotation:
      x: 0.0
      y: 0.0
      z: 7.377698243047212e-14
      w: 1.0
---
```

So you'd just need a [relay](https://github.com/ros-tooling/topic_tools?tab=readme-ov-file#relay) from /robot1/tf -> /tf and /robot1/tf_static -> /tf_static and it should be good.

dkavroulakis, 2025-01-07:
Thank you very much, I used a relay for namespace robot and it worked! 
More specific, I ran this command: ros2 run topic_tools relay namespace/tf tf

MoffKalast, 2025-01-07:
Glad I could help 👍 

Interestingly there was [another discussion on this very topic](https://discourse.ros.org/t/tf-tree-in-a-multi-robot-setup-in-ros2/41426/5) yesterday and they raised an interesting point, that we could just subscribe to the non slash version of the topic to allow them to be namespaced. I'm not 100% sure if that would work from the roslibjs side but I'll have to test it out and see if it works reliably with and without a namespace, since it would solve this edge case by just removing two characters xd.

dkavroulakis, 2025-09-25:
Hello again, I would like to mention that I have two physical turtlebot4 lite robots. 
When I run the command `ros2 run tf2_tools view_frames`, I get: No tf data received.
When I run this command: `ros2 run tf2_tools view_frames --ros-args -r __ns:=/robot1 -r /tf:=tf -r /tf_static:=tf_static `
or `ros2 run topic_tools relay robot1/tf tf`  and `ros2 run topic_tools relay robot1/tf_static tf_static`, I get this file:

[frames_2025-09-25_12.58.46.pdf](https://github.com/user-attachments/files/22534322/frames_2025-09-25_12.58.46.pdf)

pondersome, 2025-09-29:
>>P.S. Is that a Pioneer 3-AT? In 2025? 😁

i somehow missed that comment... why yes, yes it is! the beasts are remarkably durable. got mine via auction from the usaf. the onboard electronics were solid. my first ros2 project was to migrate one of the original ros packages to ros2.