MoffKalast/vizanti #123: 2D Nav Goal isn't working

dkavroulakis, 2025-01-08:
I send a goal to the navigation stack using the arrow, but nothing happens. I don't know why, but when I do the same into rviz it is working fine. Any ideas to fix this?

**Server Platform(s):**
 - OS: [Ubuntu 22.04]
 - Architecture: [e.g. x86_64]
 - ROS Version: [Humble]
 - Branch/Tag: [ros2]

MoffKalast, 2025-01-08:
Hmm, my first thought would be fixed frame mismatch, at least that sometimes causes issues. 

Check if that is set to the same one as in Rviz (in [Global Settings](https://github.com/MoffKalast/vizanti/wiki/Feature-List)), since some nav setups tend to expect that to be the map frame and the default is typically odom. 

If that's not it, then I'll need more info about your setup (e.g. rviz, vizanti saved configs).

dkavroulakis, 2025-01-08:
Thank you very much!!! The problem was in Global settings, it has odom as default value and when I choosed map option it worked.

MoffKalast, 2025-01-08:
Nice, yeah it's a common pitfall with nav2 in general I suppose. In cases like this it's often practical to check with topic echo and comparing what's the difference in the messages being sent. :wink: