MoffKalast/vizanti #133: Still can't get waypoints working

ValkyrX, 2025-02-14:
Hi M8, sorry to reopen this issue but I have tried and tried to get waypoints working on the athena2 robot, its the last thing I need to sort out, on the other ticket I opened you linked 2 python scripts, I have tried both of those, the second one I actually saw the data for the waypoint come from the web page but the Athena does nothing with it, I realise its probably the Athena robot not getting the info but im a little out of my depth here , can you help plz ?

Thanks alot
K

MoffKalast, 2025-02-14:
Well does the Athena's navigation stack accept more than one goal at a time? The waypoints widget currently does Path and PoseArray messages, so one of those would need to be converted into the type of message it accepts, assuming there is a ROS subscriber for it.

Do you have any examples from slamtec on how they expect goals to be sent to it? Does sending single goals from rviz work? It would be easiest if you could compile all the info you've got on their end of the setup.

ValkyrX, 2025-02-14:
So if I run RVIZ I can indeed simply use simple/goal and it moves to the point I select, also in your vizanti software if I use the 2D nav goal that also works so that's what's confusing me , if I can send a nav goal using rviz and your software I don't see why its ignoring the ones selected in waypoints. What I can see however is that in the topic list on waypoints widget it auto selects slanted_ros_sdk_server_node/global_path_plan (Path) and it says topic status is ok? would of thought if it works using simple nav goal then the waypoint nav stack should also work or am I being to simplistic ?

ValkyrX, 2025-02-14:
added a few screenshots here to show the setup

[image]
[image]

ValkyrX, 2025-02-14:
This is what it shows on 2d nav pointer , this works fine only difference I can see is the topic its subscribed to, I can't change the topic on waypoints at all 

[image]

MoffKalast, 2025-02-15:
Hmm yeah [looking at their spec for the sdk_server_node](https://developer.slamtec.com/docs/slamware/ros-sdk-en/2.8.2_rtm/slamware_ros_sdk_server_node/) it looks like the Path is just a publisher for showing where the robot will drive, which is fairly typical.

One option is using the /move_base_simple/goal, which would require a node that subscribes to waypoints and then sends each one to the robot, when it detects that the robot is within range of the previous one.

There's also the move_to_locations ([slamware_ros_sdk/MoveToLocationsRequest](https://developer.slamtec.com/docs/slamware/ros-sdk-en/2.8.2_rtm/msgs/move_to_locations_request/)) topic that would be better if happens to work since it should do the goal checking internally for each point. 

Can you try triggering it in the terminal first, with something like this:

```bash
rostopic pub /move_to_locations slamware_ros_sdk/MoveToLocationsRequest "locations:
- {x: 1.0, y: 0.0, z: 0.0}
- {x: 1.0, y: 1.0, z: 0.0}
- {x: 0.5, y: 0.5, z: 0.0}
options:
  opt_flags: 0
  speed_ratio: { value: 0.5 }
yaw: 1.57"
```

This would in theory send some points for the robot to drive to, to see what happens.

ValkyrX, 2025-02-15:
so not sure I did this right, see screenshot :-) I just get an error 

[image]

MoffKalast, 2025-02-15:
There's some issue with the options variable, try doing this: paste only `rostopic pub /move_to_locations slamware_ros_sdk/MoveToLocationsRequest` and then press tab twice so it autocompletes the rest, that should write out the correct format.

MoffKalast, 2025-02-15:
Maybe it needs more than one point to work? The value: 0.0 for speed_ratio could also be setting the max speed to zero. If you have any line to slamtec support it would be helpful if they have any examples on how to get that working. Hopefully it wasn't just a wishful thinking kind of API that they intended to implement later...

ValkyrX, 2025-02-15:
I can try but they take weeks to reply to anything as they r in china :-) , its just odd it works with the 2D arrow , im wondering if there is any way to make the waypoints widget use the simple goal setting as I know that works, but that's beyond my knowledge to setup .. I suppose it would have to send the goal, then wait till it gets there and then send the next , hmm , there is also an api that you can set goto command with 
{
  "action_name": "slamtec.agent.actions.MoveToAction",
  "options": {
    "target": {
      "x": 0,
      "y": 0,
      "z": 0
    },
    "move_options": {
      "mode": 0,
      "flags": [],
      "yaw": 0,
      "acceptable_precision": 0,
      "fail_retry_count": 0,
      "speed_ratio": 0
    }
  }
}

these work setup as python scripts , that's how I did the navigation buttons on the web interface, you click the button and the web server runs a php scripts pointed to the python scripts, that works well ..

MoffKalast, 2025-02-16:
Ok then it's probably not the velocity that's the problem, but it's likely just broken internally somewhere.

Welp given that this is a constant pain for lots of people I suppose I'll put together a short demo node that takes in waypoints and publishes simple goals so we can put it to rest.

ValkyrX, 2025-02-17:
Ok soooo, its further along , what happens is this, when I plan a route and I press send it, the robot moves totes first point but immediately errors showing the below screenshot, I think its something to do with the map topic having a /   nearly there :-)

[image]

ValkyrX, 2025-02-17:
ignore ignore last , its working I simply needed to change the fixed frame :-) yay its working great , ty ty :-)

ValkyrX, 2025-02-17:
Ok one last thing I promise lol, is there any way to get the waypoints mission to start by pressing a button like the send it but on the button widget ? I know I can emergency stop from it but can't see if I can start from it ?

Thanks

MoffKalast, 2025-02-17:
Ah glad it's finally working 😁 

> get the waypoints mission to start by pressing a button like the send it but on the button widget 

Yeah I've been thinking about that, currently there's no option and there's [some confusion regarding short and long presses](https://github.com/MoffKalast/vizanti/issues/127) to even open it.

So far the best idea I've had is to make it only use single taps, but then open a dropdown menu instead, which would list something like:
- Send
- Stop
- Edit XY
- Edit Z
- Configure

Then it would also be easier to integrate [this Z value addition](https://github.com/MoffKalast/vizanti/discussions/96) for airplanes and submarines without having to tap multiple times.

And then if that works well enough, change a few other widgets to that system as well.

ValkyrX, 2025-02-17:
That sounds like a plan :-) any way that means you dont have to keep opening the menu would be awesome makes it less clunky but well impressed with the software its exactly what we needed.  Regarding other widgets for my needs can't really think of anything else that we need, a speed dial on the main screen instead of up top would be a nice to have , also a widget that can be setup to move pan and tilt on the webcam instead of the base would be good as well but that I presume would need to be on a case by case as all cams use different controls I guess.

MoffKalast, 2025-02-17:
Yeah that might be a thing at some point, I'm slowly working on adding an IMU widget which would optionally render a large navball and the same could be copied over to the speed widget too.

I still think it should be possible to get the pantilt mapped to the joystick without many issues, if you manage to get a working python script example that makes the needed API calls for moving it in all directions I could really easily wrap that into a working node, much like that battery and temperature converter.

ValkyrX, 2025-02-18:
So thats annoying, apparently there is no API to the cams controls :-( booo , maybe i should look at a diferent wifi ip camera that has a built in web http service then maybe that would work ...

MoffKalast, 2025-02-18:
Ah haha, I was gonna ask which model of PTZ webcam it is for my own use since that sort of open functionality is rather unheard of over usb, but yeah I guess you might have to find an IP cam instead, those might have more readily available REST APIs but it's still not that common afaik. 

There's also the option of stacking two servos and calling it a day, or getting a wider lens 😄 

Anyhow, I think we can close this thread for now and you can open a new discussion if you find one that could be potentially integrated. It's certainly something that would be pretty handy to have.

closed 2025-02-18 by MoffKalast

MoffKalast, 2025-02-23:
@ValkyrX Hey about the dropdown menu I've just implemented it in https://github.com/MoffKalast/vizanti/pull/112. The Z value thing is still experimental and the whole thing needs porting to ROS 2, so it'll probably be a few weeks before it's actually merged but if you want to give it a test in the meantime, I'd be glad to have some feedback 😉

Looks a bit like so and is indeed a lot more practical:
[image]

ValkyrX, 2025-02-23:
Hey m8, that sounds good, what branch is it in ? noetic devel. Or ?

Kurt Ingleson
Engineering Manager
***@***.******@***.***>
[signature_3261123581]<https://emailrtc.voiceflex.com/?c=704a7bd758>
https://pacomms.co.uk<https://pacomms.co.uk/>

[Title: Twitter - Description: Twitter icon]<https://twitter.com/> [Title: Facebook - Description: Facebook icon] <https://www.facebook.com/>  [Title: LinkedIn - Description: LinkedIn icon] <https://www.linkedin.com/>  [Title: Google Plus - Description: Google Plus icon] <https://plus.google.com/>  [Title: Youtube - Description: Youtube icon] <https://www.youtube.com/>
[Title: Hexagon bar - Description: Hexagon bar]

Did you know ? if you have a support request the fastest way to report it is via our ticketing system available at http://helpdesk.pacloud.online<http://helpdesk.pacloud.online/>

This e-mail and any attached files are confidential and may be legally privileged. If you are not the addressee, any disclosure, reproduction, copying, distribution, or other dissemination or use of this communication is strictly prohibited. If you have received this transmission in error please notify the sender immediately and then delete this mail.
E-mail transmission cannot be guaranteed to be secure or error free as information could be intercepted, corrupted, lost, destroyed, arrive late or incomplete, or contain viruses. The sender therefore does not accept liability for any errors or omissions in the contents of this message which arise as a result of e-mail transmission or changes to transmitted date not specifically approved by the sender.
If this e-mail or attached files contain information which do not relate to our professional activity we do not accept liability for such information
This email has been scanned by the PA Communications Cloud Email Network.

image.png (view on web)<https://github.com/user-attachments/assets/c8c8948c-d621-4a9b-92c3-8529a962bf34>

image.png (view on web)<https://github.com/user-attachments/assets/c8c8948c-d621-4a9b-92c3-8529a962bf34>

MoffKalast, 2025-02-23:
noetic-waypointz2 is the #112 PR branch.

ValkyrX, 2025-02-25:
Just an fyi, the waypoints addition is working well on my system (ROS Noetic) , looking forward to more widgets :-)

MoffKalast, 2025-02-26:
Hmm well it's entirely possible that some case is not handled right in the path to simple goal node, though it would be hard to fix without a solid error log. Worst case would be just the node crashing though, it shouldn't bring down the whole system.