MoffKalast/vizanti #140: Battery widget not displaying charging status

ValkyrX, 2025-03-11:
So i am using the latest noetic_devel, i am running the battery.py topic separate for now to get the battery widget working, (the one you created) its publishing data on topic /battery , i can see on that topic the messages are charging when its on dock charging and off dock , on dock = 1 off dock = 3. However in the battery widget charging shows as ? , any ideas ?

MoffKalast, 2025-03-11:
Ha funny you should mention that, I've noticed this recently when upgrading an old robot, there's an ID conflict between the widget status indicator and the charge status indicator on the frontend. 

The #138 fixes it for noetic and #139 for humble/jazzy, plus it adds a new set of icons to display a lightning when charging. There's also a few minor fixes for the waypoints widget. I'm planning on testing it a bit more until I merge it, but feel free to try it out (noetic-battery-fix branch).

ValkyrX, 2025-03-11:
yep that's fixed it :-) awesome I can see when its charging now and be sure it is .. Any progress on that gimbal ball widget you mentioned ? just wondering :-) no pressure lol

MoffKalast, 2025-03-12:
Ah good! I'll close this once it's merged. Got a bit sidetracked on the gimbal, I'm still not entirely satisfied with how it renders. But I do have an interesting quadcopter project coming up where it might be useful so it's probably getting added at some point.

> no pressure lol

Clearly 😄

closed 2025-04-11 by MoffKalast