MoffKalast/vizanti #149: Call_service InvalidServiceException: Service /rosapi/topics does not exist

dkavroulakis, 2025-05-09:
Hello, sometimes I get this error on the terminal when I launch the vizanti project:
[rosbridge_websocket-1] [ERROR] [1746779002.975197132] [vizanti_rosbridge]: [Client da87d53b-b442-4517-bc5d-16799c769c66] [id: call_service:/rosapi/topics:5] call_service InvalidServiceException: Service /rosapi/topics does not exist
and I don't know why. Any ideas to fix this?

**Server Platform(s):**
 - OS: [Ubuntu 22.04]
 - Architecture: [x64]
 - ROS Version: [Humble]
 - Branch/Tag: [ros2]

MoffKalast, 2025-05-09:
Hmm that does sound vaguely familiar, though more in the broken dev build kind of situation. Might also be DDS problem.

Are you running with rosbridge or rws? Rosdbridge has a separate rosapi node which might be crashing, rws does it all internally so it would have to be gone entirely. I would suggest trying the other one and checking if it persists or if the error is any different.

dkavroulakis, 2025-05-12:
I am running with rosbridge! I use FastDDS and export RMW_IMPLEMENTATION=rmw_fastrtps_cpp, because I have turtlebot4 lite robot and I follow this page: [Turtlebot4 lite setup](https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html). Is it possible to run RWS Backend?

MoffKalast, 2025-05-12:
For RWS with FastDDS you'd need to switch [to the dynamic version of it](https://github.com/MoffKalast/vizanti/tree/ros2?tab=readme-ov-file#recommended---rws-backend), which is otherwise identical but has a few extra features (topic introspection which rws uses) and can interface with other machines using the default version. But yes it should work.

I think I do recall that there was a time when rosapi was broken on ROS 2 in rosbridge but that was fixed a year or two ago, so it could be an old install on your end I suppose. I would try doing an `apt update` and `apt install ros-humble-rosbridge-suite` in to make sure it's latest.

Btw, you haven't really mentioned what kind of problems this causes. I presume if rosapi's not working then the browser frontend doesn't fully load or doesn't show topics? And that it happens only sometimes makes me think it could be related to system load or something like that. Is there any correlation with subscribing to lots of topics in the browser and this happening or is it just random?

dkavroulakis, 2025-05-12:
Yes, you are right, it happens only sometimes and it could be related to system load or something like that. If rosapi is not working then the browser frontend doesn't fully load and doesn't show topics, so it doesn't show anything, no map, no costmap etc. So, I will try with rws later and get back to you. Thank you very much!

MoffKalast, 2025-06-27:
Ah what do you know, I've ran into just this issue today with a misconfigured cyclone setup:

```bash
[rosbridge_websocket-1] [INFO] [1751051158.059537227] [vizanti_rosbridge]: Rosbridge WebSocket server started on port 5001
[rosbridge_websocket-1] [INFO] [1751051158.546643564] [vizanti_rosbridge]: Calling services in existing thread
[rosbridge_websocket-1] [INFO] [1751051158.549380494] [vizanti_rosbridge]: Sending action goals in existing thread
[rosbridge_websocket-1] [INFO] [1751051158.551723044] [vizanti_rosbridge]: Client connected. 1 clients total.
[rosbridge_websocket-1] [INFO] [1751051158.574736107] [vizanti_rosbridge]: Client disconnected. 0 clients total.
[rosbridge_websocket-1] [INFO] [1751051158.644509643] [vizanti_rosbridge]: Calling services in existing thread
[rosbridge_websocket-1] [INFO] [1751051158.645243868] [vizanti_rosbridge]: Sending action goals in existing thread
[rosbridge_websocket-1] [INFO] [1751051158.646258045] [vizanti_rosbridge]: Client connected. 1 clients total.
[rosbridge_websocket-1] [INFO] [1751051158.705782313] [vizanti_rosbridge]: [Client f6f61fd5-87b4-46fc-b5c1-51233af168b6] Subscribed to /vizanti/tf_consolidated
[rosbridge_websocket-1] [INFO] [1751051158.707439472] [vizanti_rosbridge]: [Client f6f61fd5-87b4-46fc-b5c1-51233af168b6] Subscribed to /tf_static
[rosbridge_websocket-1] [ERROR] [1751051158.708343283] [vizanti_rosbridge]: [Client f6f61fd5-87b4-46fc-b5c1-51233af168b6] [id: call_service:/rosapi/nodes:3] call_service InvalidServiceException: Service /rosapi/nodes does not exist
[rosbridge_websocket-1] [ERROR] [1751051158.709321471] [vizanti_rosbridge]: [Client f6f61fd5-87b4-46fc-b5c1-51233af168b6] [id: call_service:/vizanti/list_packages:4] call_service InvalidServiceException: Service /vizanti/list_packages does not exist
[rosbridge_websocket-1] [ERROR] [1751051158.710285045] [vizanti_rosbridge]: [Client f6f61fd5-87b4-46fc-b5c1-51233af168b6] [id: call_service:/rosapi/topics:5] call_service InvalidServiceException: Service /rosapi/topics does not exist
[rosbridge_websocket-1] [ERROR] [1751051158.711928459] [vizanti_rosbridge]: [Client f6f61fd5-87b4-46fc-b5c1-51233af168b6] [id: call_service:/vizanti/bag/status:6] call_service InvalidServiceException: Service /vizanti/bag/status does not exist
[rosbridge_websocket-1] [ERROR] [1751051158.719843755] [vizanti_rosbridge]: [Client f6f61fd5-87b4-46fc-b5c1-51233af168b6] [id: call_service:/rosapi/topics:7] call_service InvalidServiceException: Service /rosapi/topics does not exist
[rosbridge_websocket-1] [ERROR] [1751051158.720812769] [vizanti_rosbridge]: [Client f6f61fd5-87b4-46fc-b5c1-51233af168b6] [id: call_service:/rosapi/topics:8] call_service InvalidServiceException: Service /rosapi/topics does not exist
[rosbridge_websocket-1] [ERROR] [1751051158.721686771] [vizanti_rosbridge]: [Client f6f61fd5-87b4-46fc-b5c1-51233af168b6] [id: call_service:/rosapi/topics:9] call_service InvalidServiceException: Service /rosapi/topics does not exist
[rosbridge_websocket-1] [ERROR] [1751051158.742483861] [vizanti_rosbridge]: [Client f6f61fd5-87b4-46fc-b5c1-51233af168b6] [id: call_service:/rosapi/topics:10] call_service InvalidServiceException: Service /rosapi/topics does not exist
[rosbridge_websocket-1] [ERROR] [1751051158.743375354] [vizanti_rosbridge]: [Client f6f61fd5-87b4-46fc-b5c1-51233af168b6] [id: call_service:/rosapi/topics:11] call_service InvalidServiceException: Service /rosapi/topics does not exist
[rosbridge_websocket-1] [ERROR] [1751051158.754560682] [vizanti_rosbridge]: [Client f6f61fd5-87b4-46fc-b5c1-51233af168b6] [id: call_service:/rosapi/topics:12] call_service InvalidServiceException: Service /rosapi/topics does not exist
```

The fix was to add a proper `CYCLONEDDS_URI` config, here's [some info about that](https://iroboteducation.github.io/create3_docs/setup/xml-config/). 

So something like `export CYCLONEDDS_URI='/home/moffkalast/cyclonedds.xml'` in the .bashrc which would look similar to this, at least for a localhost only setup in my case:
```bash
<CycloneDDS>
   <Domain>
     <General>
        <Interfaces>
          <NetworkInterface name="lo" />
        </Interfaces>
        <DontRoute>true</DontRoute>
    </General>
   </Domain>
</CycloneDDS>
```

I'll close it for now as a DDS issue, feel free to reopen if it doesn't fix it on your end.

closed 2025-06-27 by MoffKalast