MoffKalast/vizanti #153: Running Vizanti in Docker crashes

tiko5000, 2025-05-28:
I tried to run vizanti in docker with:
- `docker build -f docker/Dockerfile -t vizanti:2.0 . --build-arg ROS_VERSION=humble`
- `docker run --rm -it --net=host --name vizanti-ros2 -e ROS_DOMAIN_ID=$ROS_DOMAIN_ID -e RMW_IMPLEMENTATION=$RMW_IMPLEMENTATION -e USE_RWS=false -v /dev/shm:/dev/shm vizanti:2.0`

It leads to a temporary freeze after `Rosbridge WebSocket server started on port 5001` followed by processes dying.
This is the full output I get in the terminal:

```
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rosbridge_websocket-1]: process started with pid [59]
[INFO] [rosapi_node-2]: process started with pid [61]
[INFO] [server.py-3]: process started with pid [63]
[INFO] [tf_consolidator-4]: process started with pid [65]
[INFO] [service_handler.py-5]: process started with pid [67]
[tf_consolidator-4] [INFO] [1748411162.032879515] [vizanti_tf_handler_node]: TF handler ready.
[server.py-3] [INFO] [1748411162.184722155] [vizanti_flask_node]: Default widget config set to /home/vizanti/colcon_ws/install/vizanti_server/share/vizanti_server/public/assets/default_layout.json
[server.py-3] [INFO] [1748411162.187038815] [vizanti_flask_node]: Flask server running at http://0.0.0.0:5000
[server.py-3] [INFO] [1748411162.189021731] [vizanti_flask_node]: Public directory set as /home/vizanti/colcon_ws/install/vizanti_server/share/vizanti_server/public/
[service_handler.py-5] [INFO] [1748411162.213336222] [vizanti_service_handler_node]: Service handler ready.
[rosbridge_websocket-1] [WARN] [1748411162.236992358] [vizanti_rosbridge]: The 'default_call_service_timeout' parameter is currently set to 0.0, which means service calls will block indefinitely if no response is received. Please note that in the Jazzy and later releases, the default value for this parameter will be updated to 5.0 seconds.
[rosbridge_websocket-1] [WARN] [1748411162.237292028] [vizanti_rosbridge]: The 'call_services_in_new_thread' parameter is currently set to False, which means service calls will block the main thread. Please note that in the Jazzy and later releases, the default value for this parameter will be updated to True.
[rosbridge_websocket-1] [WARN] [1748411162.237472519] [vizanti_rosbridge]: The 'send_action_goals_in_new_thread' parameter is currently set to False, which means sending action goals will block the main thread. Please note that in the Jazzy and later releases, the default value for this parameter will be updated to True.
[rosbridge_websocket-1] [INFO] [1748411162.239269900] [vizanti_rosbridge]: Rosbridge WebSocket server started on port 5001
[server.py-3] [WARNING] [2025-05-28 05:46:05,659] [waitress]: Task queue depth is 1 
[server.py-3] [WARNING] [2025-05-28 05:46:05,660] [waitress]: Task queue depth is 2 
[server.py-3] [WARNING] [2025-05-28 05:46:05,716] [waitress]: Task queue depth is 1 
[server.py-3] [WARNING] [2025-05-28 05:46:05,724] [waitress]: Task queue depth is 1 
[server.py-3] [WARNING] [2025-05-28 05:46:05,725] [waitress]: Task queue depth is 1 
[server.py-3] [WARNING] [2025-05-28 05:46:05,732] [waitress]: Task queue depth is 1 
[server.py-3] [WARNING] [2025-05-28 05:46:05,732] [waitress]: Task queue depth is 2 
[server.py-3] [WARNING] [2025-05-28 05:46:05,740] [waitress]: Task queue depth is 1 
[server.py-3] [WARNING] [2025-05-28 05:46:05,742] [waitress]: Task queue depth is 2 
[server.py-3] [WARNING] [2025-05-28 05:46:05,751] [waitress]: Task queue depth is 1 
[server.py-3] [WARNING] [2025-05-28 05:46:05,752] [waitress]: Task queue depth is 1 
[server.py-3] [WARNING] [2025-05-28 05:46:05,762] [waitress]: Task queue depth is 1 
[rosbridge_websocket-1] [INFO] [1748411165.916174511] [vizanti_rosbridge]: Calling services in existing thread
[rosbridge_websocket-1] [INFO] [1748411165.916784232] [vizanti_rosbridge]: Sending action goals in existing thread
[rosbridge_websocket-1] [INFO] [1748411165.919814177] [vizanti_rosbridge]: Client connected. 1 clients total.
[rosbridge_websocket-1] [INFO] [1748411166.505740672] [vizanti_rosbridge]: [Client c8aad67f-0e0f-425a-bfee-06af0e014a3f] Subscribed to /vizanti/tf_consolidated
[rosbridge_websocket-1] [INFO] [1748411166.507893429] [vizanti_rosbridge]: [Client c8aad67f-0e0f-425a-bfee-06af0e014a3f] Subscribed to /tf_static
[rosbridge_websocket-1] [INFO] [1748411171.311854431] [vizanti_rosbridge]: Client disconnected. 0 clients total.
[ERROR] [rosapi_node-2]: process has died [pid 61, exit code -9, cmd '/home/vizanti/colcon_ws/install/rosapi/lib/rosapi/rosapi_node --ros-args -r __node:=rosapi'].
[ERROR] [server.py-3]: process has died [pid 63, exit code -9, cmd '/home/vizanti/colcon_ws/install/vizanti_server/lib/vizanti_server/server.py --ros-args -r __node:=vizanti_flask_node --params-file /tmp/launch_params_ppf12_ts --params-file /tmp/launch_params_x_78v4ae --params-file /tmp/launch_params_2vr0jjdq --params-file /tmp/launch_params_qwsl5m3p --params-file /tmp/launch_params_1v1yy6el --params-file /tmp/launch_params_u1j_bd40 --params-file /tmp/launch_params_729qvcd_'].
[ERROR] [tf_consolidator-4]: process has died [pid 65, exit code -9, cmd '/home/vizanti/colcon_ws/install/vizanti_cpp/lib/vizanti_cpp/tf_consolidator --ros-args -r __node:=vizanti_tf_handler_node'].
[ERROR] [rosbridge_websocket-1]: process has died [pid 59, exit code -9, cmd '/home/vizanti/colcon_ws/install/rosbridge_server/lib/rosbridge_server/rosbridge_websocket --ros-args -r __node:=vizanti_rosbridge --params-file /tmp/launch_params_not07zqi --params-file /tmp/launch_params_9__3jvjr --params-file /tmp/launch_params_s6yr1bir --params-file /tmp/launch_params__cg26mey --params-file /tmp/launch_params_ds3_vfu4 --params-file /tmp/launch_params_51aqdcxe --params-file /tmp/launch_params_sthhvhp7 --params-file /tmp/launch_params_tw3xqmed --params-file /tmp/launch_params_cy8o3ly_'].
[ERROR] [service_handler.py-5]: process has died [pid 67, exit code -9, cmd '/home/vizanti/colcon_ws/install/vizanti_server/lib/vizanti_server/service_handler.py --ros-args -r __node:=vizanti_service_handler_node'].
```

**Server Platform(s):**
 - OS: [e.g. Ubuntu 22.04]
 - Architecture: [e.g. x86_64]
 - ROS Version: [e.g. Humble]
 - Branch/Tag: [e.g. ros2]

**Client Platform(s):**
 - OS: [e.g. Ubuntu 22.04]
 - Architecture: [e.g. x86_64]
 - Browser: [e.g. Chrome]

MoffKalast, 2025-05-28:
Hmm I've just tested it with those params on a 24.04 machine without ROS and it seems to build and run fine, even if the $ROS_DOMAIN_ID and $RMW_IMPLEMENTATION are unset which I'd half expect to just crash, I guess it uses internal defaults from the osrf image.

I'll see if I can test on 22.04 tomorrow but given that it's docker I really wouldn't expect it to be any different. I'm not really that much of an expert on Docker in general though, @samkys any ideas on what to try?

tiko5000, 2025-06-03:
Cannot reproduce at the moment because docker build currently fails due to https://github.com/osrf/docker_images/issues/807 with 

 ```
> [15/16] RUN cd /home/vizanti/colcon_ws && rosdep install -i --from-path src -y ;:
1.744 Reading package lists...
2.238 Building dependency tree...
2.334 Reading state information...
2.454 The following NEW packages will be installed:
2.454   ros-humble-control-msgs
2.653 0 upgraded, 1 newly installed, 0 to remove and 482 not upgraded.
2.653 Need to get 432 kB of archives.
2.653 After this operation, 8070 kB of additional disk space will be used.
2.653 Err:1 http://packages.ros.org/ros2/ubuntu jammy/main amd64 ros-humble-control-msgs amd64 4.6.0-1jammy.20240728.221104
2.653   404  Not Found [IP: 64.50.233.100 80]
2.660 E: Failed to fetch http://packages.ros.org/ros2/ubuntu/pool/main/r/ros-humble-control-msgs/ros-humble-control-msgs_4.6.0-1jammy.20240728.221104_amd64.deb  404  Not Found [IP: 64.50.233.100 80]
2.660 E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
2.664 ERROR: the following rosdeps failed to install
2.664   apt: command [apt-get install -y ros-humble-control-msgs] failed
2.667 executing command [apt-get install -y ros-humble-control-msgs 
```

tiko5000, 2025-06-04:
Maybe an important info to add. I don't have ROS2 installed on my system. 
I use ROS2 inside other docker containers and want to visualize that data using vizanti.
But the other docker container is not running when i encounter that issue.

Managed to build the Dockerfile with temporarily adding this before `apt-get update`:
```
# Replace invalid key due to https://github.com/osrf/docker_images/issues/807
RUN rm /etc/apt/sources.list.d/ros2-latest.list \
  && rm /usr/share/keyrings/ros2-latest-archive-keyring.gpg

RUN export ROS_APT_SOURCE_VERSION=$(curl -s https://api.github.com/repos/ros-infrastructure/ros-apt-source/releases/latest | grep -F "tag_name" | awk -F\" '{print $4}') ;\
    curl -L -s -o /tmp/ros2-apt-source.deb "https://github.com/ros-infrastructure/ros-apt-source/releases/download/${ROS_APT_SOURCE_VERSION}/ros2-apt-source_${ROS_APT_SOURCE_VERSION}.$(. /etc/os-release && echo $VERSION_CODENAME)_all.deb" \
    && apt-get update \
    && apt-get install /tmp/ros2-apt-source.deb \
    && rm -f /tmp/ros2-apt-source.deb
```
I double checked and I have a clean clone of the ros2 branch and no other containers are running when I try to run this. Still getting this Errors:

```
docker run --rm -it --net=host --name vizanti-ros2 -e ROS_DOMAIN_ID=$ROS_DOMAIN_ID -e RMW_IMPLEMENTATION=$RMW_IMPLEMENTATION -e USE_RWS=false -v /dev/shm:/dev/shm vizanti:2.0
[INFO] [launch]: All log files can be found below /home/vizanti/.ros/log/2025-06-04-08-55-21-661196-cni-till-58
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rosbridge_websocket-1]: process started with pid [59]
[INFO] [rosapi_node-2]: process started with pid [61]
[INFO] [server.py-3]: process started with pid [63]
[INFO] [tf_consolidator-4]: process started with pid [65]
[INFO] [service_handler.py-5]: process started with pid [67]
[tf_consolidator-4] [INFO] [1749027321.793019373] [vizanti_tf_handler_node]: TF handler ready.
[server.py-3] [INFO] [1749027321.969323573] [vizanti_flask_node]: Default widget config set to /home/vizanti/colcon_ws/install/vizanti_server/share/vizanti_server/public/assets/default_layout.json
[server.py-3] [INFO] [1749027321.972332586] [vizanti_flask_node]: Flask server running at http://0.0.0.0:5000
[server.py-3] [INFO] [1749027321.973682888] [vizanti_flask_node]: Public directory set as /home/vizanti/colcon_ws/install/vizanti_server/share/vizanti_server/public/
[service_handler.py-5] [INFO] [1749027321.985682212] [vizanti_service_handler_node]: Service handler ready.
[rosbridge_websocket-1] [WARN] [1749027323.934774208] [vizanti_rosbridge]: The 'default_call_service_timeout' parameter is currently set to 0.0, which means service calls will block indefinitely if no response is received. Please note that in the Jazzy and later releases, the default value for this parameter will be updated to 5.0 seconds.
[rosbridge_websocket-1] [WARN] [1749027323.935136299] [vizanti_rosbridge]: The 'call_services_in_new_thread' parameter is currently set to False, which means service calls will block the main thread. Please note that in the Jazzy and later releases, the default value for this parameter will be updated to True.
[rosbridge_websocket-1] [WARN] [1749027323.935326003] [vizanti_rosbridge]: The 'send_action_goals_in_new_thread' parameter is currently set to False, which means sending action goals will block the main thread. Please note that in the Jazzy and later releases, the default value for this parameter will be updated to True.
[rosbridge_websocket-1] [INFO] [1749027323.944916847] [vizanti_rosbridge]: Rosbridge WebSocket server started on port 5001
[ERROR] [rosapi_node-2]: process has died [pid 61, exit code -9, cmd '/home/vizanti/colcon_ws/install/rosapi/lib/rosapi/rosapi_node --ros-args -r __node:=rosapi'].
[ERROR] [service_handler.py-5]: process has died [pid 67, exit code -9, cmd '/home/vizanti/colcon_ws/install/vizanti_server/lib/vizanti_server/service_handler.py --ros-args -r __node:=vizanti_service_handler_node'].
[ERROR] [server.py-3]: process has died [pid 63, exit code -9, cmd '/home/vizanti/colcon_ws/install/vizanti_server/lib/vizanti_server/server.py --ros-args -r __node:=vizanti_flask_node --params-file /tmp/launch_params_t8bup790 --params-file /tmp/launch_params_3b6y_oxb --params-file /tmp/launch_params_tdkxyn6y --params-file /tmp/launch_params_tw1aw67y --params-file /tmp/launch_params_bvsc58lc --params-file /tmp/launch_params_jc_ext26 --params-file /tmp/launch_params_ieoqhmaq'].
[ERROR] [tf_consolidator-4]: process has died [pid 65, exit code -9, cmd '/home/vizanti/colcon_ws/install/vizanti_cpp/lib/vizanti_cpp/tf_consolidator --ros-args -r __node:=vizanti_tf_handler_node'].
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[rosbridge_websocket-1] registered capabilities (classes):
[rosbridge_websocket-1]  - <class 'rosbridge_library.capabilities.advertise.Advertise'>
[rosbridge_websocket-1]  - <class 'rosbridge_library.capabilities.publish.Publish'>
[rosbridge_websocket-1]  - <class 'rosbridge_library.capabilities.subscribe.Subscribe'>
[rosbridge_websocket-1]  - <class 'rosbridge_library.capabilities.defragmentation.Defragment'>
[rosbridge_websocket-1]  - <class 'rosbridge_library.capabilities.advertise_service.AdvertiseService'>
[rosbridge_websocket-1]  - <class 'rosbridge_library.capabilities.call_service.CallService'>
[rosbridge_websocket-1]  - <class 'rosbridge_library.capabilities.service_response.ServiceResponse'>
[... cut ...]
[rosbridge_websocket-1]  - <class 'rosbridge_library.capabilities.action_feedback.ActionFeedback'>
[rosbridge_websocket-1]  - <class 'rosbridge_library.capabilities.action_result.ActionResult'>
[rosbridge_websocket-1]  - <class 'rosbridge_library.capabilities.send_action_goal.SendActionGoal'>
[rosbridge_websocket-1]  - <class 'rosbridge_library.capabilities.unadvertise_action.UnadvertiseAction'>
[rosbridge_websocket-1] Exiting due to SIGINT
[INFO] [rosbridge_websocket-1]: process has finished cleanly [pid 59]
```

tiko5000, 2025-06-04:
Another Finding: 
When is Set ROS_DOMAIN_ID=0 or ROS_DOMAIN_ID= docker crashes.
ROS_DOMAIN_ID>=1 docker does not crash.

samkys, 2025-06-07:
On this note: **"But i seems to be an issue within our network. When I disable my Network connection, the Docker Container does not crash anymore."**

@MoffKalast would know the available internal server arguments better than my self. From the container side, before trying to debug the nodes further, I think it's worth debugging your network since that made a difference. Do you have another machine on your network running ROS2? One thing we've done in the past is isolated computers. Try setting the localhost only variable and see if it helps: https://docs.ros.org/en/foxy/Tutorials/Beginner-CLI-Tools/Configuring-ROS2-Environment.html#the-ros-localhost-only-variable

If you need robot-to-robot comms, we'll need to revert back but the above will at least be a starting point to see it stops crashing on a single robot.

MoffKalast, 2025-06-07:
Hmm yeah I think in terms of ROS related connections that will be all, there's also the waitress static serve used by the flask node, which [would need to be set to debug separately](https://docs.pylonsproject.org/projects/waitress/en/latest/logging.html), but out of all nodes, I think that's the only one that doesn't crash so my bet is more on the DDS sockets being blocked by something as @samkys suggests.

Could maybe try to attach a debugger to the DDS directly (it would have to be specific to fast or cyclone or whichever you're using I suppose) or check what's using which ports directly with `sudo ss -tulnp` or similar and compare the before and after launching.